Emotiv/cortex-example #125: Opt-out unavailable

viswajithkn, 2021-02-05:
create session result 
 {
    "error": {
        "code": -32155,
        "message": "The opt-out feature is unavailable."
    },
    "id": 1,
    "jsonrpc": "2.0"
}

I am getting the above issue, though our license has the opt-out set. What could be the possible reasons?

tungntEmotiv, 2021-02-08:
Hi @viswajithkn ,

Could you run API to get the current optOut described in doc (https://emotiv.gitbook.io/cortex-api/records/configoptout) and show us the result?

Thanks

bchaperon, 2021-02-08:
Hello @viswajithkn 

Do you mean that you get this error when you call the method "createSession"?
https://emotiv.gitbook.io/cortex-api/session/createsession

As far as I know, the only way to get this error is to call the method "configOptOut".
https://emotiv.gitbook.io/cortex-api/records/configoptout

Can you call the method configOptOut with the status "get" and show me the response?
```
{
  "id": 1,
  "jsonrpc": "2.0",
  "method": "configOptOut",
  "params": {
    "cortexToken": "xxx",
    "status": "get"
    }
}
```
You should get a response like this one:
```
{
  "id": 1,
  "jsonrpc": "2.0",
  "result": {
    "currOptOut": false,
    "forceOptOut": false,
    "message": "Get opt-out configuration successfully",
    "optOutAvail": true
  }
}
```
If the field "optOutAvail" is **false** then your application doesn't have the permission to set the opt-out.
In that case, trying to set the opt-out (with the status "set") will return the error -32155.

You should also call the method "getLicenseInfo" to make sure that your application is using the right license.
https://emotiv.gitbook.io/cortex-api/authentication/getlicenseinfo

If your application is using to right license, but you cannot set the opt-out, then maybe the config of your license in the Emotiv cloud is incorrect. In that case, you have to contact the [Emotiv customer support](https://www.emotiv.com/contact-support/).

viswajithkn, 2021-02-08:
Thanks for your response. We got the license fixed. But now at the very end if we try to export the record we get an error message with code -32137 and "message": "Cortex cannot open the record data file." 

export record result 
 {
    "id": 13,
    "jsonrpc": "2.0",
    "result": {
        "failure": [
            {
                "code": -32137,
                "message": "Cortex cannot open the record data file.",
                "recordId": "b8e53066-2524-4d85-a0dc-3e7aaf8eaac9"
            }
        ],
        "success": []
    }
}
Why is cortex not able to open the record data file?

bchaperon, 2021-02-09:
Hi @viswajithkn 

This is probably because the output folder doesn't exist.
You must create the folder before you call "exportRecord". If the folder doesn't exist, then the export will fail with error 32137.
https://emotiv.gitbook.io/cortex-api/records/exportrecord

viswajithkn, 2021-02-09:
I do have the export folder available. Thanks

bchaperon, 2021-02-09:
A possible explanation is that Cortex doesn't have the permission to write in your folder.
Can you try to write in a different folder? 
The temp folder of your OS should work. For example, on macOS, try "/tmp".
You can also try your home folder.

Another possible explanation is that the name of the record contains a special character that is not valid in a filename. In theory, Cortex is able to detect such characters and it automatically replaces them with an underscore. But maybe we missed a few cases.
Can you try to export a record with a simple name, like "foo"?

Also, if you create and export a record from EmotivPRO, does it work?

closed 2021-02-09 by viswajithkn