Emotiv/cortex-example #126: examples train.py error -32002 "Invalid License Key."

ivaylov, 2021-02-15:
I have BCI API Free (Lite FREE) License today I did try record.py and train.py I get this error. m I doing something wrong with clientId and clientSecret? I did try registering new apps but the result is the same Invalid License Key. These python examples are pro (paid) version?
 
 
query headset --------------------------------
EPOCPLUS-########
connect headset --------------------------------
connect headset result {
    "id": 2,
    "jsonrpc": "2.0",
    "result": {
        "command": "connect",
        "message": "The headset EPOCPLUS-######## has been connected or is connecting."
    }
}
request access --------------------------------
{
    "id": 3,
    "jsonrpc": "2.0",
    "result": {
        "accessGranted": true,
        "message": "The user has granted access right to this application."
    }
}
authorize --------------------------------
auth request
 {
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": "#myclientIdfromcortexapps",
        "clientSecret": "#myclientSecretfromcortexapps",
        "license": "free",
        "debit": 100
    },
    "id": 4
}
auth result
 {
    "error": {
        "code": -32002,
        "message": "Invalid License Key."
    },
    "id": 4,
    "jsonrpc": "2.0"
}
Traceback (most recent call last):
  File "train.py", line 101, in <module>
    t=Train()
  File "train.py", line 11, in __init__
    self.c.do_prepare_steps()
  File "cortex.py", line 190, in do_prepare_steps
    self.authorize()
  File "cortex.py", line 122, in authorize
    self.auth = result_dic['result']['cortexToken']
KeyError: 'result'

bchaperon, 2021-02-17:
Hi @ivaylov 

When you call "authorize", I see that you set the parameter "license" to "free".
This is incorrect. The value of this parameter should be the id of the license, not its name.
https://emotiv.gitbook.io/cortex-api/authentication/authorize

I recommend you to completely remove this parameter, because in most cases, Cortex will find your license automatically.

ivaylov, 2021-02-17:
Thank you so I will try.

bchaperon, 2021-03-03:
If you have a paid license, then I think you received the license id by email when you subscribed.

closed 2021-03-03 by bchaperon