Emotiv/cortex-example #127: 32004

grosssj, 2021-02-19:


bchaperon, 2021-02-19:
Hi @grosssj 

To open a session with a headset, the status of the headset must be "connected". If the status is "discovered", then you must call "controlDevice" to connect the headset.
https://emotiv.gitbook.io/cortex-api/session/createsession

If you run Emotiv BCI in parallel, then BCI will call "controlDevice" to connect the headset, so your application don't need to do it. That's why it works in that case.

The error code -32019 means that you tried to activate a session when the “localQuota” of your license is zero. You can check the "localQuota" by calling the method "getLicenseInfo". 
https://emotiv.gitbook.io/cortex-api/authentication/getlicenseinfo

To solve this error, you must increase this local quota. You do this by setting the parameter "debit" of the method "authorize". For example, you can call "authorize" with a "debit" of 50, and this will increase your local quota by 50. Then you will be able to create and activate 50 sessions on your computer.

Please note that if you only want use the BCI features of the API, then you don't need to activate the session. You can call "createSession" with the status "open", instead of "active". And so you don't need to worry about the debit and the local quota.

grosssj, 2021-02-19:
Hi bchaperon,

thank you very much for the feedback. As we do not currently have a license for more than BCI activities, I look forward to taking advantage of your information when I get up in the morning. Thank you very much.  I have been trying to get the example to function as a template for doing BCI interfacing to control an object in real time. Thanks again. Is there a good reference on the api (doc or code) that is more complete than the api document referenced in the downloaded files for the example code?  Any pointers would be greatly appreciated. Thanks again. Stay safe.

closed 2021-03-03 by bchaperon