Emotiv/cortex-example #130: authorize error after successful requestAccess

jyauri, 2021-04-13:
I have an **authorize** error in my registered CortexApp with EEG. It gives me 

{'error': {'code': -32002, 'message': 'Invalid License Key.'}, 'id': 1, 'jsonrpc': '2.0'}

but the requestedAccess is successful:

{'id': 1, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The user has granted access right to this application.'}}

To test what is happening, I registered another CortexApp but without checking EEG, and it passes and gives a cortex_token, but this is useless since my work is to develop an app to access and process EEG data.

How can I get the cortex_token for an application which requires EEG access?

bchaperon, 2021-04-15:
Hi @jyauri 

To get the EEG data from the API, you need a paid PRO license. Then you have to fill a form on the EMOTIV web site. The links for the forms are in the documentation:
https://emotiv.gitbook.io/cortex-api/#license

If you already did these steps, then try to call the method "authorize" without sending any license ID. In most cases, Cortex will find your license ID automatically, based on your EMOTIV id. So you don't need this parameter.
```
{
    "id": 1,
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": "xxx",
        "clientSecret": "xxx",
        "debit": 1
    }
}
```

jyauri, 2021-04-15:
Hi @bchaperon 

Now, we have a EmotivPro Bussiness licence. Since I am a researcher student I fill the form, but I din't receive any response, despite the email "We should respond to your request within 24 hours".

To use the CortexApi, additionally should we have to pay this license [Business API License](https://www.emotiv.com/product/business-sdk-license/ )?

bchaperon, 2021-04-19:
Hi @jyauri 

If you didn't get any response to your request, then please contact the EMOTIV customer support. Give them the EMOTIV id that owns the PRO Business license and they will be able to sort it out for you.
https://www.emotiv.com/contact-support/

jyauri, 2021-04-21:
@bchaperon , sorry I am late, Until today I didn't receive any response,  I have to contact to the support, thank you?

bchaperon, 2021-04-22:
@jyauri Yes please, contact the EMOTIV support. Your issue is clearly related to your license, so the support will be able to help you.

jyauri, 2021-04-27:
No solution, and I am disappointed with Emotiv.  After a week, they say me that even the free BCI option allows me to access data, such as power spectrum and others, but noting technical answer about my problem. Only, ask me to register in the form, which I did many times, and registered many apps to get the cliente_id, and client_secret, none of them works.

When I call the "method": "authorize" never get me the TOKEN to be used later, instead it show me the error 
{'error': {'code': -32002, 'message': 'Invalid License Key.'}, 'id': 1, 'jsonrpc': '2.0'}

I am disappointed to have bought this headset, perhaps with another headset it could have been easier to access the data in real time.

bchaperon, 2021-04-28:
Hi @jyauri 

I just talked to the support team. They will contact you again very soon.

closed 2021-04-28 by bchaperon