Emotiv/cortex-example #131: The name of clientID and clientSecret does not exist in the current context

chenlixiao-a, 2021-04-15:
I creat a APP and get the clientID and clientSecret. But when I run it it tells me that"The name of clientID and clientSecret does not exist in the current context."Can you tell me how to deal with it?
`[Header("Connection")]
    public string socketUrl = "wss://localhost:6868";

    [Header("Emotiv")]
    public string clientId= WiPiYYURLcBSKviJ4mxYA498H4IgNnh7YPuOBgB0;
    public string clientSecret= LDiO1XONHUtr0Mv88K8eJMxFNKKN9deUUV1J9jrbfvFoWvkRQihYWFZaLfUTwgXmTuRCswodn5XxgkBnEvFLJIQQpLMVdOf6VjikHVE9cTBKtKrDbpbZJQ4X1mzjDrVH;
    public string headsetId = "EPOCPLUS-3B9AEC58";`

bchaperon, 2021-04-15:
Hi @chenlixiao-a 

Please, do not publish the client secret on your application on a public forum. The client secret must remain a secret :)

About your error, how did you get it? Does it come from the Cortex API? What message did you send to Cortex to receive this error?

chenlixiao-a, 2021-04-15:
Hello,
I want to run the project  [https://github.com/BrandlMax/BrainFramework].I follow the instructions.I l get the clientID and clientSecret. Then I added them in the code . After that I click the  the button of "play" in Unity.The errors tells me that "The name of xxxxxxxx(my clientID and clientSecret) does not exist in the current context."

bchaperon, 2021-04-15:
If the error comes from the BrainFramework, then please contact the authors of this software to get help.
EMOTIV is not involved in the development of this software and so we cannot provide any support for it.

closed 2021-04-19 by bchaperon