Emotiv/cortex-example #132: Unity getContactQuality is always -1

floriangamillscheg, 2021-04-18:
Hello! 

I'm having difficulties getting the Unity example to work. When I start the Unity example, it **correctly authorizes** it, but when the "**ensure good contact quality" screen** appears it just shows for 1 frame the correct quality, 
[image]
but then the quality is always 0.
[image]

I printed the value that I get from the DataStream::**getContactQuality** method and it is always **-1**, except when the screen initially appears where it seems to work for a fraction of a second, but then it is always -1... In the DevDataBufferMethod::GetContactQuality() method a comment says `//return -1 if the moment there is no battery come.` What does this mean? I charged the battery to 100%... 
I would be very happy if somebody could help me, since I'm writing my bachelor thesis, in which I use Emotiv Insight and therefore understanding how the plugin in works by using the example would be **very** helpful.

tungntEmotiv, 2021-04-22:
Hi @floriangamillscheg ,

We have just updated unity-plugin and unity-example to support for Emotiv Cortex 2.7.0. Please update and check again. Let us know about result. Thanks

floriangamillscheg, 2021-04-24:
HI @tungntEmotiv , 

yes thank you, it now works! I have 2 other questions:
1. is there a way to decrease the time between the measurement of the performance metrics data? It seems the performance data is only updated about every 30 seconds. Is there a way to decrease this time?
2. Is the profile only used for the mental commands and facial expression? Or in other words, can a trained profile help achieve more accurate data in the performance metrics?

Thanks!

bchaperon, 2021-04-26:
Hi @floriangamillscheg 

1. The API should send you the PM data every 10 seconds. Internally, Cortex updates the PM every 0.5 seconds, but you need a paid PRO license with the option "High-Res Performance Metrics" to see the PM at 2 Hertz.
https://www.emotiv.com/emotivpro/

2. Yes, the profile is only for mental command and facial expressions. You cannot train the performance metrics.

closed 2021-05-05 by bchaperon