Emotiv/cortex-example #136: A function for output string variables.

yarac12, 2021-04-30:
In the examples named live_advance.py, train.py, train_advance.py...etc. (using the python codes), the function .live() was being used to output the trained commands.
Is there any function we can use that returns mental commands and facial expressions as string variables??

bchaperon, 2021-05-05:
Hi @yarac12 

Sorry, I don't understand your question. What do you want to do exactly?

yarac12, 2021-05-05:
Instead of streaming the data to the prompt I want to save the data inside variables. so I can use these variables later to connect them with arduino.

yarac12, 2021-05-05:
instead of this function ( in python )
	def live(self, profile_name):
		print('begin live mode ----------------------------------')
		# load profile
		status = 'load'
		self.c.setup_profile(profile_name, status)

		# sub 'com' stream and view live mode
		stream = ['com', 'fac']
		self.c.sub_request(stream)

 instead of stream function, we want to have result= "something", meaning if we print result we want to get neutral or lift or push as a word.

NguyenDuyToan, 2021-05-06:
@yarac12  you could modify the ` sub_request` method under cortex.py module. The method simply waits for stream data from the websocket connection and print out the data values. So basically you could store the stream data to a variable instead of printing it.

closed 2021-05-12 by bchaperon