Emotiv/cortex-example #140: Problem with the connection between the Cortex API and Flutter

nicolaspgamba, 2021-05-12:
Hello, I am currently trying to make a connection between my App in Flutter and Cortex API through the websocket, as such it does not generate an error, but I am not receiving a response.

I am sending a json as a request (as the documentation indicates). The Websocket should respond with a json, but I am not receiving any response, it generates a null.

[image]

[image]

[image]

bchaperon, 2021-05-19:
Hi @nicolaspgamba 

I have no experience with Flutter, and we don't use this framework at EMOTIV. So it impossible for me to check if your source code is correct or not.

Which OS are you using? 
Can you try the Python example (or any other example) from this Github repo and see if it works?
Does your program handle the errors that may occur when you connect to the web socket?

nicolaspgamba, 2021-05-24:
Hi @bchaperon 

It seems that the problem is with the port, Flutter tells me that it has problems trying to certify it. Does the API only work on that port? Is there no way to change it?

This is the error that generates me
[image]

bchaperon, 2021-05-25:
Hi @nicolaspgamba 

The problem comes from the certificate. The Cortex web socket server uses a self-signed certificate. By default, most web socket clients consider that such certificate is invalid.

Maybe you can configure Flutter to accept self signed certificate. I know it is possible in other languages.
I see that other people had the same issue with Flutter: https://github.com/dart-lang/sdk/issues/34284

closed 2021-06-02 by bchaperon

shathaalotaibi, 2022-01-17:
Did the connection work with you?