Emotiv/cortex-example #146: Invalid License Key

CarlosValleA, 2021-09-01:
Hi, I'm following the example sub_data.py and I can't pass the authorization step.

I keep getting the following error:

```python
auth result 
 {
    "error": {
        "code": -32002,
        "message": "Invalid License Key."
    },
    "id": 4,
    "jsonrpc": "2.0"
}

```

The license field was left with `""`  as described in the documentation.  My client_id is `5GTv46GwkKC510SwnmPirPfzpnpadlig75tczT0Z` and user jamaro93. Also I'm using the EPOC-X 14 channels.

Thanks

tungntEmotiv, 2021-09-01:
Hi @CarlosValleA ,

Did you put a valid license at user object in sub_data.py ?

user = {
    "license" : "your emotivpro license, which could use for third party app", # put your license here
    "client_id" : "your client id",
    "client_secret" : "your client secret",
    "debit" : 100
}

CarlosValleA, 2021-09-01:
Hi @tungntEmotiv,  the license key is empty as `""` as it is [not required ](https://emotiv.gitbook.io/cortex-api/authentication/authorize#examples) according to the documentation.  The user variable is:

```python
user = {
    "client_id" : "5GTv46GwkKC510SwnmPirPfzpnpadlig75tczT0Z",
    "client_secret" : "CLIENT_SECRET",
    "debit" : 100,
    "license": ""
}
```

tungntEmotiv, 2021-09-01:
Hi @CarlosValleA ,

 Did you tick to "My App require EEG access" box when you create the Cortex App ? Because you don't have a eeg access license. So you shouldn't tick the box. With the basic license you can subscribe other data streams such as "dev", "met","com, "fac"
You can reference to the [link](https://www.youtube.com/watch?v=eEWSUBxPYwI&t=316s)

CarlosValleA, 2021-09-01:
Yes, I tick the "My App require EEG access" field and it says YES. I have an emotivPRO Student, is that not enought to use the API ?. Also the error is generated in line 137 in `cortex.py`  while waiting for the answer of the method `authorize`, and it dosen't reach the point where the suscribe request is reached.

tungntEmotiv, 2021-09-06:
Hi @CarlosValleA ,

If you have EmotivPRO student -> you will have EEG access  but  you haven't submitted application form at  https://www.emotiv.com/student-sdk-application-form/ to get access for Cortex API for EEG data. 
Thanks

closed 2021-10-15 by bchaperon