Emotiv/cortex-example #151: Invalid license key after purchasing it

JalSs, 2021-11-01:
Hi, I'm having problem with the licensing where it still show invalid after I bought the Student Pro license. During generating the cortex id and secret id, I tick the box since I need the use of EEG access in my app but it still showing error. I already fill the Student SDK form and mentioning it in the Emotiv contact Support but still no response in my email. It have been many days. My error is similar to this: https://github.com/Emotiv/cortex-v2-example/issues/130#issue-857104116

I hope that the Emotiv team can help me in this situation, thank you in advance for the cooperation.
Emotiv Id: fypbca
email: fypbca@gmail.com

tungntEmotiv, 2021-11-02:
Please try again @JalSs . Let us know if the issue still happen. Thanks

JalSs, 2021-11-02:
The issue already solved, thank you for your concern :)

closed 2021-11-02 by JalSs