Emotiv/cortex-example #152: How To Connect with Cortex API?

shathaalotaibi, 2021-11-09:
Is it mandatory to have the Emotive Launcher running, to be able to call wss://localhost:6868?

bchaperon, 2021-11-10:
Hi @shathaalotaibi 

You must install the EMOTIV Launcher and run it once to login with your EMOTIV id. After that, you don't have to run it to use the Cortex API.
Installing the Launcher also installs the CortexService, which is a background process. When you connect to wss://localhost:6868 you actually connect to the CortexService, you don't connect to the Launcher application.

shathaalotaibi, 2021-11-10:
I am developing an Android application and I cannot download the Emotiv launcher on it, is there another way?

shathaalotaibi, 2021-11-10:
I am developing an Android application and I can not download the Emotiv launcher on it, is there another way?

bchaperon, 2021-11-11:
On Android you must install the application "EMOTIV App". But this app is currently in Beta.
The only way to get it is to send a request to the EMOTIV customer support.
https://www.emotiv.com/contact-support/

closed 2021-11-17 by bchaperon