Emotiv/cortex-example #155: python eeg logger

areejhameed, 2021-11-17:
hi..
I'm using emotiv insight , I need to perform online EEG classification but the python code for eeglogger doesn't work, can you help me in this issue?
thanks

bchaperon, 2021-11-18:
Hi @areejhameed 

Can you give me more details please? Which Python script are you running exactly? And what error do you get?

areejhameed, 2021-11-18:
Hi..
Thank you for your response..
Previously,I used emotiv epoc + with the attached file code and an error
with can not loading the (edk.dll) library then exit from the code..but now
i'm using emotiv insight (5 channels) with record.py code but in the online
classification this will delay the processing , i need to manipulate the
data from the file directly to get the raw eeg signals as the case of the
attached eeg logger code .
thanks
 eegloggerالاصلية.py
<https://drive.google.com/file/d/17B6Gmez3sv6wZg-c8xW_JczO3PHc7INf/view?usp=drivesdk>

bchaperon, 2021-11-19:
Hi @areejhameed 

The edk.dll is the old Emotiv SDK. It is not supported anymore.
Instead, you should download and install our latest software "Emotiv Launcher". 
https://www.emotiv.com/my-account/downloads/

This will also install the Cortex API on your computer. The documentation for this API is here:
https://emotiv.gitbook.io/cortex-api/

Then you can run the Python examples from this repository.
https://github.com/Emotiv/cortex-v2-example/tree/master/python

areejhameed, 2021-11-19:
Thank you so much

closed 2021-11-29 by bchaperon