Emotiv/cortex-example #156: Problem exporting record

enrique-argones, 2021-11-23:
Hi,

We are developing an app to capture an EEG database. We need to perform the captures while the subjects perform certain operations / interactions. We create the record following the procedures shown in the cortex-v2-example, and everything works apparently just fine, but when we ask the service to export the record to a given local folder, we get an error code and the export fails. I attach here the complete trace of the program as a initial step.
[App Log.txt](https://github.com/Emotiv/cortex-v2-example/files/7586952/App.Log.txt)

nhungpham11, 2021-11-25:
Hi,
I see that you called API exporting record before data file closed.
Pls wait until you receive the warning:
` warning  {"jsonrpc":"2.0","warning":{"code":18,"message":{"message":"The session data has been successfully saved."}}}`
After that you can call to export record successfully.

enrique-argones, 2021-11-25:
Thank you very much, I will implement this and let you know whether this fixes the issue.

enrique-argones, 2021-11-25:
Indeed, this fixed the problem. I was confused, since I could not see the warning being processed in the example where the record is exported. I had to add a signal to the cortexclient class that emits the warning so I can detect it. Now, it works.

Thanks!

nhungpham11, 2021-11-26:
Thank for your feedback and nice to hear that your problem has been resolved :) .
Can you tell me what language of Example that you followed. 
We will check it and add missing point.
Thanks!

enrique-argones, 2021-11-26:
I am developing in C++ using Qt. The patch I made consists of adding a signal in the CortexClient that emits the warning code and message, so other classes can capture it.

Thank you.

closed 2021-12-06 by bchaperon