Emotiv/cortex-example #160: RAW eeg monitoring with EPOC

Batuhanyucel, 2022-01-05:
I need to do RAW eeg monitoring with python using cortex API or something else like LSL. Do you have the ready code or what steps ı should follow can you help me ?
 İf I need Pro licence for this project ı will have it.
(bybatuhanyucel@gmail.com)

tungntEmotiv, 2022-01-07:
Hi @Batuhanyucel ,

To get EEG data you need a PRO license . You can reference the python example for EEG subscribing at https://github.com/Emotiv/cortex-v2-example/blob/master/python/sub_data.py

Thanks

Batuhanyucel, 2022-01-09:
[cid:c1f23b8f-8278-4df8-a530-b9b87ee4a817]
Thank you for your feedback. What ı'm gonna write to licence part can you help me ? İs it my order number ?
I have emotiv pro licence already
________________________________
Gönderen: tungntEmotiv ***@***.***>
Gönderildi: 7 Ocak 2022 Cuma 05:42
Kime: Emotiv/cortex-v2-example ***@***.***>
Bilgi: Batuhanyucel ***@***.***>; Mention ***@***.***>
Konu: Re: [Emotiv/cortex-v2-example] RAW eeg monitoring with EPOC (Issue #160)

Hi @Batuhanyucel<https://github.com/Batuhanyucel> ,

To get EEG data you need a PRO license . You can reference the python example for EEG subscribing at https://github.com/Emotiv/cortex-v2-example/blob/master/python/sub_data.py

Thanks

Batuhanyucel, 2022-01-09:
I got my licence ID , emotiv ID and ssecret then ı'm trying to subscribe data and ı'm getting 
ERROR : "code": -32021,
        "message": "Invalid Client Credentials."

I have pro licence but the API does'nt let me work.

tungntEmotiv, 2022-01-11:
Hi @Batuhanyucel ,

Did you register your application  and put the clientId and secret to user ?
`
user = {
    "license" : "your emotivpro license, which could use for third party app",
    "client_id" : "your client id",
    "client_secret" : "your client secret",
    "debit" : 100
}
`
The video https://www.youtube.com/watch?v=oKvhDe7tNts guide training example but data subscribing is similar.

Batuhanyucel, 2022-01-11:
i did it and if ı tried not eeg required apps they are working. But if i mark "my app will be required to raw eeg" ı'm getting  Invalid Client Credentials error  (I already have a  pro licence )

Batuhanyucel, 2022-01-11:
if you want to check ı can give to you a my all ID's with mail .

tungntEmotiv, 2022-01-11:
Could you send me the log lines on your console ? The error said that the clientId or secret is invalid. my email is tungnguyen@emotiv.com

Batuhanyucel, 2022-01-11:
I sended you my user code lines  with my cortex app page .
Thank you for interest a lot

Batuhanyucel, 2022-01-13:
I sended my cortex.txt with e-mail.
For your information.
Batuhan YÜCEL

tungntEmotiv, 2022-01-13:
Yes, as the error said that the clientId and secret is invalid.
Can I set up a teamviewer for your machine?  Where are you located? We will
choose  a suitable time  for us.

Batuhanyucel, 2022-01-13:
I'm living istanbul and ı'm availablle for 4-5 hours .  Today You can join my teamwiever whenever you want in 5 hours. İf u avaliable too.
I gave to you my teamwiever ID 's from e-mail

Batuhanyucel, 2022-01-13:
Thank you for your high interest and concern. You solved my problem with your high skill and knowledge.

closed 2022-01-13 by Batuhanyucel