Emotiv/cortex-example #162: The type of data

Batuhanyucel, 2022-01-17:
I am getting real time data with cortex sub.data py but ı can’t understand what is the type of this data? 
what should ı do for reachimg  voltage values from these datas  
[image]

bchaperon, 2022-01-17:
Hi @Batuhanyucel 
The format of the data you receive from the Cortex API is described here:
https://emotiv.gitbook.io/cortex-api/data-subscription/data-sample-object

Batuhanyucel, 2022-01-17:
İt says “ For each EEG sensor, you get 1 value in microvolt” but values are over 400 mV it is real ?

Batuhanyucel, 2022-01-17:
Correction : over4000

bchaperon, 2022-01-18:
Hi @Batuhanyucel 
Yes, the values usually are around 4200 micro volts. It is expected.
But before you do analyze the data, it is recommended that you remove the DC offset.
https://emotiv.gitbook.io/emotivpro-v3/notes-on-the-data/dc-offset

closed 2022-01-18 by Batuhanyucel