Emotiv/cortex-example #168: Cortex Python Unsubscribe

chowa3, 2022-02-28:
What is the best way to unsubscribe from a data stream? There is not an unsubscribe method in the Python Cortex API. I tried the following: 

self.c = Cortex(user, debug_mode=True)
self.c.close_session()

However, this does not work for my application as I need to later re-open the session. When attempting to: 
server.c.create_session(server.c.auth, server.c.headset_id)

"behavior": "Cortex has stopped all the subscriptions of session
[image]

chowa3, 2022-02-28:
Disregard! I figured it out. I added the following code to cortex.py

# Added by chowa3
    def unsub_request(self, streams):
        print('unsubscribe request --------------------------------')

        unsub_request_json = {
            "id": 1,
            "jsonrpc": "2.0",
            "method": "unsubscribe",
            "params": {
                "cortexToken": self.auth,
                "session": self.session_id,
                "streams": [streams]
            }
        }

        self.ws.send(json.dumps(unsub_request_json))

tungntEmotiv, 2022-02-28:
Yes, thanks for your contribution.

closed 2022-03-03 by bchaperon