Emotiv/cortex-example #169: Problem with authorization

CarlosA-Mtz, 2022-03-01:
Greetings, I bought the student license for Emotiv pro and the following issue appears:
ine 137, in authorize
    self.auth = result_dic['result']['cortexToken']
KeyError: 'result'

tungntEmotiv, 2022-03-03:
Hi @CarlosA-Mtz ,

Which python example did you run ? 
Please help to set debug_mode=True in the __init__ function as below. We need more log to debug
`
    def __init__(self):
        """
        Constructs cortex client and bind a function to handle subscribed data streams for the Train object
        If you do not want to log request and response message , set debug_mode = False. The default is True
        """
        self.c = Cortex(user, debug_mode=True)
`
and make sure you fill  below user object
`user = {
    "license" : "", // set empty string here 
    "client_id" : "your client id",
    "client_secret" : "your client secret",
    "debit" : 100
}
`
Please note that when you run the example firs time, you might need to approve your application via Emotiv Launcher.

CarlosA-Mtz, 2022-03-06:
I followed the instructions as they appear in the video, last time the same error happened to me but they told me that I needed the license, which I already got

tungntEmotiv, 2022-03-07:
Hi @CarlosA-Mtz ,

Could you send us your emotivID and the clientId via email tungnguyen@emotiv.com ? We will check more

tungntEmotiv, 2022-03-14:
Thanks @CarlosA-Mtz ,
We have received your email. Because you create an application which require EEG access. So you need request API access via https://www.emotiv.com/cortex-sdk-application-form/

closed 2022-03-23 by bchaperon