Emotiv/cortex-example #171: Cortex has unloaded the profile

Motlib97, 2022-03-03:
Greetings
I'm trying to build an my application on python. I've loaded my profile using the 'setupProfile'  method and I'm getting a stream of mental commands data. Now, when I rerun the code I get this message 'A profile is already loaded for this headset. You must unload it first.'. 
I know how to unload the profile, but I don't want to do that every time I rerun the code!!
Any help is appreciated ..

{'error': {'code': -32127, 'message': 'A profile is already loaded for this headset. You must unload it first.'}, 'id': 7, 'jsonrpc': '2.0'}
{'com': ['neutral', 0.0], 'sid': 'f6fdf5cb-6261-48a0-b331-0a76ad0935de', 'time': 1646325175.5681}
{'com': ['neutral', 0.0], 'sid': 'f6fdf5cb-6261-48a0-b331-0a76ad0935de', 'time': 1646325175.6931}
{'com': ['neutral', 0.0], 'sid': 'f6fdf5cb-6261-48a0-b331-0a76ad0935de', 'time': 1646325175.8181}
{'com': ['neutral', 0.0], 'sid': 'f6fdf5cb-6261-48a0-b331-0a76ad0935de', 'time': 1646325175.9431}
{'com': ['neutral', 0.0], 'sid': 'f6fdf5cb-6261-48a0-b331-0a76ad0935de', 'time': 1646325176.0681}
{'jsonrpc': '2.0', 'warning': {'code': 15, 'message': {'behavior': 'Cortex has unloaded the profile', 'headset': 'EPOCX-D80B1EC7', 'profile': 'motlib'}}}
{'com': ['neutral', 0.0], 'sid': 'f6fdf5cb-6261-48a0-b331-0a76ad0935de', 'time': 1646325176.1931}

bchaperon, 2022-03-07:
Hi @Motlib97 

Before you load a profile, you should call "getCurrentProfile" to check which profile is already loaded.
https://emotiv.gitbook.io/cortex-api/bci/getcurrentprofile

If your profile is already loaded, then you don't need to load it again.
If another profile is loaded, then you must unload it before loading your profile.
If no profile is loaded, then you can load your profile immediately.

Motlib97, 2022-03-11:
oh I see, thank you

closed 2022-03-14 by bchaperon