Emotiv/cortex-example #179: Stuck at executing https://github.com/Emotiv/cortex-v2-example/blob/master/python/sub_data.py

vishwasdalal, 2022-05-28:
I do not have the license to the raw data stream, I want to access the free APIs
I made the following changes to https://github.com/Emotiv/cortex-v2-example/blob/master/python/sub_data.py

# list data streams
    streams = ['mot','met','pow', 'dev']
    s.start(streams)

My python is stuck at this
[image]

tungntEmotiv, 2022-05-30:
Hi @vishwasdalal ,

I think the problem might from missing a python lib such as python-dispatch or something wrong when receive response message from websoscket. So 

1) do you install python-dispatch via "pip install python-dispatch" ? 
2) Please add a debug log to on_message function at cortex.py as below? And let us know the result
`
def on_message(self, *args):
        recv_dic = json.loads(args[1])
        print( recv_dic)
        if 'sid' in recv_dic:
`
Thanks

vishwasdalal, 2022-05-30:
hi @tungntEmotiv 
1.do you install python-dispatch via "pip install python-dispatch" ?
- I have python-dispatch installed already
2. Please add a debug log to on_message function at cortex.py as below? And let us know the result
- well I inserted a breakpoint at **on_message** and the program is stuck before that, **on_message** is never called.

vishwasdalal, 2022-05-30:
[image]

this is where this shows up, this is in **threading.py**

closed 2022-05-31 by vishwasdalal