Emotiv/cortex-example #180: Can't login on Cortex UI

danifm1321, 2022-06-18:
When I execute cortex_code_example.js I have the following error: 
```
You must login on CortexUI before request for grant access then rerun
```
I'm using EMOTIV Launcher 3.3.0.26 on Ubuntu and I don't get any access request in the launcher. I have the TCP port 6868 listening and the user look like this:
```
  let user = {
      "clientId":clientId,
      "clientSecret":clientSecret,
      "debit":1
  }
```
with variables clientId and clientSecret with the pertinent values.

Any help? Did I misunderstand anything?

nguoithichkhampha, 2022-06-18:
@danifm1321 , it say 

> You must login on CortexUI before request for grant access then rerun

did you login your emotiv id ?

danifm1321, 2022-06-20:
I logged in the launcher (I even set up a virtual device), if that's the answer to your question.

nguoithichkhampha, 2022-06-20:
did you keep Emotiv Launcher running when you run the example code ? you should see a popup message from Emotiv Launcher ask for accept the request.

danifm1321, 2022-06-20:
Yes, Emotiv Launcher is running while I run
```
node cortex_code_example.js
```
in a terminal, and I don't get any popup in the launcher.

I am using [this](https://www.emotiv.com/download_emotiv_launcher_ubuntu.php) launcher, is it the correct one right?

danifm1321, 2022-06-20:
Instead of logging the value of `accessGanted` I logged the value of `requestAccessResult` and I got 
`{"error":{"code":-32021,"message":"Invalid Client Credentials."},"id":1,"jsonrpc":"2.0"}`.

So I hard coded clientId and clientSecret instead of getting them from files and it worked, although I checked the variables were getting the correct values.

Thank you for the support and sorry for the problems I've caused.

closed 2022-06-20 by danifm1321