Emotiv/cortex-example #184: Invalid Key licence in Python with registered Cortex app

craigvear, 2022-08-03:
Running sub.data.py. Have supplied your_app_client_id & your_app_client_secret in the main() as given in EMOTIV account.
Headset Insight2 connected, and Cortex App showing in EMOTIV launcher.

Error returns :

```
{
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": "removed for security",
        "clientSecret": "removed for security",
        "license": "",
        "debit": 10
    },
    "id": 4
}
handle_error: request Id 4
{'code': -32002, 'message': 'Invalid License Key.'}
```

Why invalid licence?

nguoithichkhampha, 2022-08-03:
@craigvear , because you send empty license in the request
`"license": ""`
if you don't know exactly your license id. you can remove the params "license", our system will choose the best match 
`{
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": "removed for security",
        "clientSecret": "removed for security",
        "debit": 10
    },
    "id": 4
}`

craigvear, 2022-08-03:
Thanks. Which line of code do I change for that ? Is it in cortex.py?

craigvear, 2022-08-04:
removed licence param in LOC449 in cortex.py. same response

```
 {
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": "removed for security",
        "clientSecret": "removed for security",
        "debit": 10
    },
    "id": 4
}
handle_error: request Id 4
{'code': -32002, 'message': 'Invalid License Key.'}
```

nguoithichkhampha, 2022-08-04:
@tungntEmotiv pls help to check this issue

craigvear, 2022-08-04:
I now seem to be getting the following:

handle_error: request Id 4
{'code': -32130, 'message': 'The cloud token is refreshing.'}

tungntEmotiv, 2022-08-05:
@craigvear ,
Do you have PRO license or free license? 
If you only have free license, please don't tick "My App required EEG access"  as below:
[image]

If you have PRO license, but still get the issue. Please send us your clientID and emotivID via email tungnguyen@emotiv.com We will check more.
Thanks

craigvear, 2022-08-05:
Have basic license. Already ticked eeg for my app.

craigvear, 2022-08-05:
working - thank you

closed 2022-08-05 by craigvear