Emotiv/cortex-example #188: Having trouble with authorization

andresnevarez, 2023-01-02:
Hi i have the basic licence and i am trying to get access from cortex in python with these parameters:

`{
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": "N0QrgbkfPrO7CpwqC74dBpRyVqqSZNd3bPOvrV0v",
        "clientSecret": "**********",
        "license": "",
        "debit": 10
    },
    "id": 4
}`

However i got a message:
{'code': -32002, 'message': 'Invalid License Key.'}

I CHECKED EEG ACCESS during setup of app.

My emotivID is: andres.nevarez and my clientID is: nN0QrgbkfPrO7CpwqC74dBpRyVqqSZNd3bPOvrV0v
Can you help me please? thank you

andresnevarez, 2023-01-02:
Had the wrong license

andresnevarez, 2023-01-02:
I already have PRO-LITE however i can't access to eeg data or facial on API. Thanks

tungntEmotiv, 2023-01-03:
Hi @andresnevarez ,
With PRO-Lite license you can not access EEG data via API . For developer, you can reference license plan here https://www.emotiv.com/developer/
Because your free license can not access to EEG data so when you CHECKED EEG ACCESS during setup of app -> got the above error. However with free license , you can access to BCI API but you need to create new one without "CHECKED EEG ACCESS" 
Please try and let us know the result. Thank you

closed 2023-03-24 by nguoithichkhampha