Emotiv/cortex-example #190: run cortex.py --ImportError: cannot import name 'Dispatcher'

Yangdudua, 2023-01-14:
i run cortex.py,but it show "ImportError: cannot import name 'Dispatcher'".Then i charge it to dispatcher and it is ok.But now it show
Traceback (most recent call last):
  File "E:\data\python\cortex-v2-example-master\python-cortex\cortex.py", line 60, in <module>
    class Cortex(dispatcher):
TypeError: module.__init__() takes at most 2 arguments (3 given)

tungntEmotiv, 2023-01-16:
Hi @Yangdudua ,

Does your project reference the latest our python example ? Could you try run sub_data.py and let us know the result ?

danielgonzalez3, 2023-04-27:
similar issue here:
`ImportError: cannot import name 'Dispatcher' from 'pydispatch' (`

maybe we can lockdown on `requirements.txt`?

danielgonzalez3, 2023-04-27:
https://stackoverflow.com/questions/70019526/importerror-cannot-import-name-dispatcher-from-pydispatch

helped me out :)

closed 2023-06-19 by bchaperon