Emotiv/cortex-example #193: Getting error on runnign sub_data.py

rajatanand86, 2023-03-15:
[Errno 2] No such file or directory
on_close
None

Hi
I am getting this error. Can anyone help?
We want to capture raw eeg data from our python script.

nguoithichkhampha, 2023-03-15:
@rajatanand86 , Can you add the message in console log when run example sub_data.py

rajatanand86, 2023-03-15:
Subscribe __init__
Subscribe __init__1
Subscribe __init__3
Subscribe __init__3
[Errno 2] No such file or directory
on_close
None

I have added print commands in def _init

nguoithichkhampha, 2023-03-16:
where do you see this log printed ? seems, this is not error log from Cortex.

rajatanand86, 2023-03-16:
The log is in cmd shell of windows. Can you guide on the Cortex app i..e where to get it?
We have purchased the API license  and have the license key and secret already added in the python code.

nguoithichkhampha, 2023-03-16:
in your project. You call script sub_data.py or use it as import file ?
Can you share your source code ?

rajatanand86, 2023-03-18:
I call sub_data.py fom command shell of windows. this is exactly the same code as in this repository and i have added the credentials.

rajatanand86, 2023-03-18:
I solved it. 

        # As default, a Emotiv self-signed certificate is required.
        # If you don't want to use the certificate, please replace by the below line  by sslopt={"cert_reqs": ssl.CERT_NONE}
        print('too')
        #sslopt = {'ca_certs': "../certificates/rootCA.pem", "cert_reqs": ssl.CERT_REQUIRED}
        sslopt={"cert_reqs": ssl.CERT_NONE}

This is from cortex.py file. I am not using any certificate for now. After running sub_data.py, i have approved the application access from launcher. But then, the below error  is coming.
code: -32022 Device limit on this license has been reached. The license on emotive pro is Emotiv PRO standard. 
Is it because in need Emotiv PRO premium??

a1848580190, 2023-03-23:
> I solved it.
>
> ```
>     # As default, a Emotiv self-signed certificate is required.
>     # If you don't want to use the certificate, please replace by the below line  by sslopt={"cert_reqs": [...]

do you solve this problem?

closed 2023-03-24 by nguoithichkhampha

rajatanand86, 2023-03-24:
The error goes when I run the python code on a machine with emotiv pro premium.