Emotiv/cortex-example #194: "error": {         "code": -32022,         "message": "Device limit on this license has been reached."

a1848580190, 2023-03-24:
This problem occurs when the same code runs on different computers

I don't understand what the device limit means.

nguoithichkhampha, 2023-03-24:
"device" mean your computer. That mean your new computer can not use the license

a1848580190, 2023-03-24:
> "device" mean your computer. That mean your new computer can not use the license

> "device" mean your computer. That mean your new computer can not use the license

This is so inconvenient, is there any way to deal with this?

a1848580190, 2023-03-24:
> "device" mean your computer. That mean your new computer can not use the license

> > "device" mean your computer. That mean your new computer can not use the license
> 
> > "device" mean your computer. That mean your new computer can not use the license
> 
> This is so inconvenient, is there any way to deal with this?

Or if I can change the device

nguoithichkhampha, 2023-03-24:
@a1848580190 , you can contact our support team to reset device for your license.

a1848580190, 2023-03-24:
> I can change the device

ok ,thanks

closed 2023-03-27 by nguoithichkhampha