Emotiv/cortex-example #195: Headset connected/disconnected status, unexpected behavior

landret, 2023-03-26:
Hi,
I found out that:
- when connecting a headset from Emotiv Launcher or Emotiv BCI, headset will appear "connected" when calling "queryHeadsets" from developper app. Which means you let people connect a headset from Emotiv apps in order to connect it in developper app.
- when disconnecting a headset from Emotiv Launcher or Emotiv BCI, headset will remain "connected" when calling "queryHeadsets" from developper app. This is inconsistent with the above behavior and unexpected. If this is the expected behavior, it should be documented.

nguoithichkhampha, 2023-03-27:
@landret , our design allows to connect headset from any app. just need to call api 
https://emotiv.gitbook.io/cortex-api/headset/controldevice
with command = "connect".

about disconnect from Emotiv Launcher or Emotiv BCI. it does not terminate the physical connection with headset. Emotiv Launcher or Emotiv BCI only call close session. So the disconnection from Emotiv apps does not affect to another app. That why when call "queryHeadsets" from developer app, you still the status is "connected".

landret, 2023-03-27:
@nguoithichkhampha thanks for the explanation. I think it would be useful to specify this in the user manual.

nguoithichkhampha, 2023-03-27:
@landret thanks for your feedback.

closed 2023-09-20 by thamha-emotiv