Emotiv/cortex-example #197: Missing warning event telling about availability of new headsets

landret, 2023-03-26:
Hi,
I am missing a warning event telling when a new headset is available (and ready to be connected). At the moment, I have to sent "queryHeadsets" every seconds to update the list of available headsets.

nguoithichkhampha, 2023-03-27:
yes, @landret . You need to call "queryHeadsets" to get available headset. in a room when there are many headsets. the event headset is available or not available may happens frequently. sending this kind of event to client app is not useful. I think it's easier to client app call "queryHeadsets" when it want to connect a headset .

landret, 2023-03-27:
@nguoithichkhampha It means a user action is required in the dev app to refresh the list with "queryHeadsets" (for instance by pressing a Refresh button). The warning event would remove this requirement. I will try to do without and see. Thanks

nguoithichkhampha, 2023-03-27:
@landret , your app can call "queryHeadsets" in background when the UI show screen to connection headset. We don't need user click a refresh button.

closed 2023-03-31 by nguoithichkhampha