Emotiv/cortex-example #198: Wireless signal returned is either 0.0 or 1.0

landret, 2023-03-31:
Hi,
the wireless signal should range between 0.0 and 1.0. However, either 0.0 or 1.0 is returned by my headsets, nothing between. 
Test done with 
- EPOC+ S/N 4A2C01C7 firmware 0x634
- Insight v1.0 S/N 5968367B firmware 0x5014

nguoithichkhampha, 2023-03-31:
@landret , how do you get wireless signal from cortex ? did you get correct order from data stream ?

landret, 2023-03-31:
@nguoithichkhampha 
Example for the insight headset: the order is : ["Battery","Signal",["AF3","T7","Pz","T8","AF4","OVERALL"],"BatteryPercent"]

Example of data sample I receive for the "dev" data stream (I am not wearing the headset here so the sensors signal quality is bad) :

{"dev":[3,0.0,[0,0,0,1,0,0],59],"sid":"09ff2962-dc7b-40f6-843e-c98691af56d9","time":1680252416.4949}

or

{"dev":[3,1.0,[0,1,0,0,0,0],64],"sid":"09ff2962-dc7b-40f6-843e-c98691af56d9","time":1680252371.4733}

nguoithichkhampha, 2023-04-03:
@landret Can you try to record data and send me an example file ? how long you see the wireless signal = 0.0. in theory, wireless signal = 0.0 mean very bad signal.

landret, 2023-04-10:
Here is a recording. 

[sample.txt](https://github.com/Emotiv/cortex-v2-example/files/11187177/sample.txt)

thamha-emotiv, 2023-09-21:
Closed it, we support in a private call.

closed 2023-09-21 by thamha-emotiv