Emotiv/cortex-example #202: use api to download the data

rajatanand86, 2023-05-23:
Hi. We are using emotiv pro premium license to use the API. The license is right now installed in our laptop A.
The moment, user uses the emotiv device, the eeg data gets uploaded to the cloud.
Now, we want to download this data to laptop B using API. How to do this? Can somebody help?
Contrary to below example, we dont have emotiv software installed in laptop B. 
https://emotiv.gitbook.io/cortex-api/records/requesttodownloadrecorddata

tungntEmotiv, 2023-05-25:
@rajatanand86 ,
You need install [Emotiv Launcher ](https://www.emotiv.com/emotiv-launcher/) on laptop B to able work with cortex service on lapTop B. After that you need to login with same account as on Laptop A. Then
- you might call queryRecords api to get list of your records
- call requestToDownloadRecordData api to download the records

rajatanand86, 2023-05-29:
Hi. Thanks for your response. I am trying to check it. and run these apis. Would there be any python codes to be able to run these apis? 
python code for downloading records is there in record.py in this repo.
But i am not able to find python code for queryRecords api.

tungntEmotiv, 2023-06-03:
Hi @rajatanand86 ,
Yes, currently, the example doesn't contain queryRecords and requestToDownloadRecord . We will update and inform you next week. In the meantime, you can do it by yourself. You might need to create a request for queryRecords at cortex.py and handle response similar to create_record . Reference to queryRecords api at https://emotiv.gitbook.io/cortex-api/records/queryrecords
Sorry for the inconvenient.
Thank you

rajatanand86, 2023-06-17:
Hello
Any luck on above? Did you get time for this?

tungntEmotiv, 2023-06-19:
Hi @rajatanand86 ,
I'm really sorry because I haven't finished updating example for the queryRecords and requestToDownloadRecord, we are busy on other projects. I will spend time to update it this week.
I have question that do you want to download records created from your application ?

rajatanand86, 2023-06-19:
Yes. For now.

rajatanand86, 2023-06-22:
Hello
I am using record.py to export the data. I am able to run till export record. and I am getting error: -32137: Cannot open the file to write. Please check your file path or permissions. I have made a folder in the desktop and have given folder: "/psaizi_temp" . but still getting error. the 20 seconfs file is getting saved in the emotiv cloud as I can see from my emotiv pro recordings. Can someone help please? Its urgent.

tungntEmotiv, 2023-06-25:
Hello @rajatanand86 ,
I have updated new example to demo how to request to download records at [support-query-record-api-python](https://github.com/Emotiv/cortex-v2-example/tree/support-query-record-api-python) branch . It is in reviewing. You can check out this branch for your work and let us know the result.
Thank you

rajatanand86, 2023-06-26:
Sure, Will check it

rajatanand86, 2023-07-11:
Hello. I am getting-32161 error code saying failed to export while data post processing is in process.
I am using the api to do the recording and save the edf file to my laptop. The recorded file is successfully showing in the Emotiv software but not getting downloaded in the laptop. 
Can someone please help?

tungntEmotiv, 2023-07-13:
Hi @rajatanand86 ,
Is The record new record or created before ? . 
Actually, after a record finished, Cortex will check and correct timestamp of record data. It will take some seconds or minutes depend of record duration. the above error code -32161 talked about that. So please try exporting after some minute after record finished

rajatanand86, 2023-07-14:
Hello. I am running the record.py file for this task and i put a time.sleep(60) liek below to prevent this error:
def export_record(self, folder, stream_types, format, record_ids,
                      version, **kwargs):
        """
        To export records
        Parameters
        ----------
        More detail at https://emotiv.gitbook.io/cortex-api/records/exportrecord
        Returns
        -------
        None
        """
        time.sleep(60)
        self.c.export_record(folder, stream_types, format, record_ids, version, **kwargs)

But the problem with this is, the recordings get done twice. For smaller recordings like 10 seconds, time.sleep() is not needed, as no -32161 error comes, but this error comes in long duration recordings.
Please help resolve this.

tungntEmotiv, 2023-07-19:
Hi @rajatanand86 ,

How long for the long record ? the timestamp correction might take some minutes for long duration record. I think you can increase the waiting timer to 2-3 minutes for long duration record

closed 2023-09-20 by thamha-emotiv