Emotiv/cortex-example #203: Warning for headset status change

landret, 2023-06-17:
Hi,
It would be great to have a warning when headset status changes. 

I am using an Emotiv Insight 1.0. When it is turned off after connecting it from my app (status goes from CONNECTED to CONNECTING), I have no way to know that it was turned off, and have to wait for 30s for the warning stating that Cortex disconnected the headset after timeout. During that time, Data Samples are no longer sent and my app users can think the GUI is frozen because nothing is happening.

hoangphamemotiv, 2023-06-19:
Hi @landret . Thank you for your suggestion. 
In some cases, Cortex doesn't know when headset turned off or it was in a poor connection. Many cases around that. 
in a case of poor connection, data can lost so much, if sending a warning for changing status, it can be spamming . 
anyways, we are thinking about this and will improve it in near future. 

[image]

landret, 2023-06-19:
Hi @hoangphamemotiv 
Thanks for your reply.

closed 2023-06-19 by hoangphamemotiv