Emotiv/cortex-example #205: Issues with Unity plugin errors -32603 and 4128

MatiHaufn, 2023-06-29:
Hello, 
I am trying to connect my Epoc X headset to Unity with the plugin. I want to use mental commands to control things in a scene. But it doesn't connect to the headset and I get some errors and I am not sure how to fix that or if I missed a setup. 

1. I changed the Config file and added the AppClientId and AppClientSecret. But should I also change something in the Authorizer (or somewhere else)? 
2. What is the cortexToken and do I need it? 
3. Console says "IsSameAppVersion: not exists directory C:Users\userAppData\Local\UnityApp\Profiles"
What does this mean and how to fix that? 
4. I have two errors 4128 and -32603 

I also checked through the DataStreamManager if it detects a headset but it doesn't, although the software is turned on and the headset connected (but maybe the issue is already one of 1.-4.).

I hope someone can help me, I would really appreciate it. 

Here is the console from Unity when I press play: 
[image]

tungntEmotiv, 2023-07-03:
Hi @MatiHaufn ,

In the unity wrapper functions , we saved some application information (app version, cortex token..) to the local machine to make the short authorizing process at next time. You can check the handling at OnGetUserLoginDone() in Authorize.cs.
The log  "IsSameAppVersion: not exists directory C:Users\userAppData\Local\UnityApp\Profiles" is normal and your application will re-authorize to get new cortex token. 
We need more information to check why you get error 4128. Please try again and then open EmotivLauncher to send the feedback , please note the clientId of your application and description of the issue in the feedback description.
Thank you

MatiHaufn, 2023-07-04:
Thank you for the fast answer. I looked at the example and tested some things and now I don't have the same issues anymore. 
It is just "code -32002 message Invalid Licence Key.method name authorize" now. But when I understand it right it is because the example is trying to read EEG data through the Pro version I don't have, right?  

Now I am just struggling with, how to read the correct Mental Commands from the headset. I am not sure about the syntax and what to do first and why. Do you have an example for that or something similar? 

Thank you for your help!

MatiHaufn, 2023-07-05:
Okay, now I am able to connect the headset and get the commands to Unity. 

The -32002 message was caused by using an App where I checked the EEG access, although I don't have the license for that. 
[image]

I cannot say exactly, how I resolved the other errors. I guess I just did something wrong in the beginning I cannot replicate anymore.  

For everyone who has also problems to start with the plugin: 
I downloaded the example from here: https://github.com/Emotiv/cortex-v2-example 
and checked the code from "SimpleExample.cs" to see how it works, because there is a possibility to see the MentalCommands through the console. 
But by downloading the example only, I had to add the plugin from here again: https://github.com/Emotiv/unity-plugin 
So I had to combine both to make it run for me. 

Thank you again for the help.

closed 2023-07-05 by MatiHaufn