Emotiv/cortex-example #206: Stream Data

emamulfarhad, 2023-08-04:
I am using sub_data.py How can I stop stream data by a keyboardInterrupt. Thank you for your time and consideration.

tungntEmotiv, 2023-08-08:
@emamulfarhad ,

Ah yes, the program's just demo how to subscribe data streams and the data streams will be received continually without stop. I think you can add a snip code by yourself to get a key from keyboard to stop the streams by call unsub() method.

shuming9886, 2023-08-24:
> I am using sub_data.py How can I stop stream data by a keyboardInterrupt. Thank you for your time and consideration.

Hello, I am using emotiv insight 2.0 and I also want to use sub_ data.py to obtain the value of foucs(foc.isActive、foc). I bought a student license for EmotivPro license, and all I need to do is apply for the Cortex SDK to run the sub_data. py? Do I need to pay any other fees? Thanks! @emamulfarhad

tungntEmotiv, 2023-08-26:
@shuming9886 ,
With your license, you can subscribe "met" data ( performance metric) but it is low rate (0.1Hz). If you want access high rate pm (2 Hz) . Please contact our CSS team. Thank you

closed 2023-09-20 by thamha-emotiv