Emotiv/cortex-example #210: "error":{"code":-32007,"message":"The session does not exist."

jrivd, 2023-09-15:
Hi, I am getting this error whenever I launch a:

```
message = {
        "id": 8,
        "jsonrpc": "2.0",
        "method": "updateSession",
        "params": {
            "cortexToken": cortex_token,
            "session": session_id,
            "status": "active"
        }
    }
```
    
There is clearly a problem with the session, becaause if I fire a  `"method": "querySessions"`, it returns empty: `Received response: {"id":1,"jsonrpc":"2.0","result":[]}.`

However, all the previous steps seem to be correct and the session opens just fine:

`Received response: {"id":4,"jsonrpc":"2.0","result":{"appId":"***","headset":{"connectedBy":"bluetooth","dfuTypes":["dfuOta"],"dongle":"0","firmware":"a11","firmwareDisplay":"0x0a11","id":"INSIGHT2-****","isDfuMode":false,"isVirtual":false,"motionSensors":["Q0","Q1","Q2","Q3","ACCX","ACCY","ACCZ","MAGX","MAGY","MAGZ"],"sensors":["AF3","T7","Pz","T8","AF4"],"settings":{"eegRate":128,"eegRes":16,"memsRate":32,"memsRes":16,"mode":"INSIGHT"},"status":"connected","virtualHeadsetId":""},"id":"*****","license":"","owner":"*****","performanceMetrics":[{"apiName":"attention","displayedName":"Attention","name":"attention","shortDisplayedName":"At"},{"apiName":"eng","displayedName":"Engagement","name":"engagement","shortDisplayedName":"En"},{"apiName":"exc","displayedName":"Excitement","name":"excitement","shortDisplayedName":"Ex"},{"apiName":"str","displayedName":"Stress","name":"stress","shortDisplayedName":"St"},{"apiName":"rel","displayedName":"Relaxation","name":"relaxation","shortDisplayedName":"Re"},{"apiName":"int","displayedName":"Interest","name":"interest","shortDisplayedName":"In"}],"recordIds":[],"recording":false,"started":"2023-09-15T15:42:07.717+02:00","status":"opened","stopped":"","streams":[]}}`

I really don't know what else to try. Thanks in advance for your support.

thamha-emotiv, 2023-09-20:
Can you try to `createSession` with `active`, so that you don't have to call `updateSession`?

jrivd, 2023-09-20:
Hi, thanks. I have tried to run directly:

```
if __name__ == "__main__":
    message = {
        "id": 4,
        "jsonrpc": "2.0",
        "method": "createSession",
        "params": {
            "cortexToken": cortex_token,        
            "headset": "INSIGHT2-A3D20705",
            "status": "active"
        }
    }

    asyncio.run(retrieve(message))
```
    
But it actually does not change much, as on the "querySessions" it still returns empty: `Received response: {"id":1,"jsonrpc":"2.0","result":[]}.`

thamha-emotiv, 2023-09-20:
@jrivd did you call `querySessions` right after you `createSession`? Please note that session is not stored anywhere after you close the session or you terminate your app, so if you call `querySessions` for previous sessions, it doesn't return anything.

jrivd, 2023-09-20:
Yes, I am indeed running this code on a jupyter notebook:

```
# request access

if __name__ == "__main__":
    message = {
        "id": 1,
        "jsonrpc": "2.0",
        "method": "requestAccess",
        "params": {
            "clientId": clientId,
            "clientSecret": clientSecret
        }
    }
    asyncio.run(retrieve(message))
    
# query headsets

if __name__ == "__main__":
    message = {
        "id": 1,
        "jsonrpc": "2.0",
        "method": "queryHeadsets"
        }

    asyncio.run(retrieve(message))    
    
    
# connect headset

if __name__ == "__main__":
    message = {
        "id": 2,
        "jsonrpc": "2.0",
        "method": "controlDevice",
        "params": {
            "command": "connect",
            "headset": "INSIGHT2-A3D20705"
        }
    }

    asyncio.run(retrieve(message))   

# authorise

if __name__ == "__main__":
    message = {
        "id": 3,
        "jsonrpc": "2.0",
        "method": "authorize",
        "params": {
            "clientId": clientId,
            "clientSecret": clientSecret,
            "debit": 10
        }
    }
    
    asyncio.run(retrieve(message))

# create session

if __name__ == "__main__":
    message = {
        "id": 4,
        "jsonrpc": "2.0",
        "method": "createSession",
        "params": {
            "cortexToken": cortex_token,        
            "headset": "INSIGHT2-A3D20705",
            "status": "active"
        }
    }

    asyncio.run(retrieve(message))

# query sessions

if __name__ == "__main__":
    message = {
        
        "id": 1,
        "jsonrpc": "2.0",
        "method": "querySessions",
        "params": {
            "cortexToken": cortex_token        

        }

    }

    asyncio.run(retrieve(message))
```

Many thanks  in advance for your support on this issue.